aktos-io/aecad #21: Please, clarify license

krasin, 2019-02-11:
First of all, thanks for developing such a promising tool!

It would be nice to know under which the license the code is released. It will help users and potential contributors to understand the limits it can be used.

Usually, people place a top-level LICENSE file, so that GitHub can automatically detect it. There are plenty of well-understood open source licenses: [Apache 2.0](https://opensource.org/licenses/Apache-2.0), [BSD](https://opensource.org/licenses/BSD-3-Clause), [MIT](https://opensource.org/licenses/MIT), [GPL](https://opensource.org/licenses/gpl-license), etc. Feel free to pick the one that suits the long-term project plans the best.

Thank you in advance.

ceremcem, 2019-02-11:
You welcome. 

The desired license is MIT. However, I'm currently not sure if there is any library we are using is conflicting with this license.

krasin, 2019-02-12:
@ceremcem thank you for clarification.

re: third-party dependencies. This is a very good point.