aktos-io/aecad #59: Run issue

nozge, 2021-01-16:
Hello i downloaded ur code and im trying to run it ... 
Can u please give me some instructions of how to set it online on a host and how to run it in localhost 
It is the same like KiCad ?

ceremcem, 2021-01-16:
> Following instructions are tested on a clean LXC container, created by:
>
>     sudo lxc-create -n aecad -t debian -B btrfs -- -r buster --packages nano sudo tmux git
>

## Requriements

* git
* tmux
* python-pip
* bash-completion (recommended)

## Preparation

1. Install `nodeenv`: 

       pip install nodeenv  # don't forget to add ~/.local/bin to your $PATH

2. Clone this repository. Assuming you are at `~/some/path`:

       git clone --recursive https://github.com/aktos-io/aecad
       cd aecad

3. Create your virtual environment:[¹](https://github.com/aktos-io/scada.js/blob/master/doc/using-virtual-environment.md): 

    > For Windows platform, follow [these instructions](https://github.com/aktos-io/scada.js/tree/master/doc/on-windows) first.

    ```console
    $ cd ./scada.js
    $ make create-venv
    $ ./venv  
    (scadajs1) $ exit 
    ````

4. Install required packages: 

    ```console
    $ cd ..   # now you are in ~/some/path/aecad
    $ make install-deps
    ```

5. Open 2 different terminals and run those commands in each of them (or skip this step completely and proceed to `#6`): 

    Terminal-1:
    ```console
    $ ./scada.js/venv
    (scadajs1) $ (cd scada.js && APP=main make development)
    ```

    Terminal-2:
    ```console
    $ ./scada.js/venv
    (scadajs1) $ (cd servers && ./run-ls webserver.ls --development)
    ```

#### 6. (Optional) Use `tmux`:

1. Add the following code to the `~/.bashrc`:

    ```sh
    # For Tmux VirtualEnv support
    tmux_get_var(){
        local key=$1
        [[ -n "$TMUX" ]] && tmux showenv | awk -F= -v key="$key" '$1==key {print $2}'
    }

    # activate the virtual environment if it is declared
    venv=$(tmux_get_var "VIRTUAL_ENV")
    if [ -n "$venv" ]; then
        source $venv/bin/activate;
    fi
    ```

2. Start the development anytime by: 

    ```sh
    ./uidev.service`
    ```

#### 7. Navigate to http://localhost:4001

# Usage

This part is under development. I can barely find time to complete the documentation. You can start with `Help` button, on the top right corner. Then simply click to green "Compile" button (`[>]`), poke around with the example. 

You can of course help me by opening issues about anything.

ceremcem, 2021-01-16:
If you don't bother using aeCAD offline, you can use it online: http://aecad.surge.sh/

ceremcem, 2021-01-16:
If you don't bother using aeCAD offline, you can use it online: https://aktos.io/aecad/ (fallback url:  http://aecad.surge.sh/)

nozge, 2021-01-16:
Hello thank you for ur reply.. i already make

1,2,3
but cannot make 4,5,6
If  u could give me more information.. it will be nice

I already installed Cygwin and temux

i dont understand it 4 5 6 ... so if u could explain it would be grateful
and how to set it online example on a webhosting ?
Or can i translate that app  thanku

ceremcem, 2021-01-16:
So you are on Windows. Please post the error you are receiving. 

Note that I can't quite remember if I tried it under Cygwin, so you may try it as exactly described in `#3`: https://github.com/aktos-io/scada.js/tree/master/doc/on-windows

> can i translate that app 

It's a huge work to do, but yes, it's doable. However, I didn't provide a translation support (mechanism) yet.

ceremcem, 2021-01-16:
Can you verify that you see those lines in `#3`: 

```console
$ ./venv  
(scadajs1) $ 
```

nozge, 2021-01-16:
> Can you verify that you see those lines in #3:
>
> $ ./venv  (scadajs1) $

I'will check it ..

I have to run it via command prompt (cmd) and then open Cygin and msys2 ?

nozge, 2021-01-16:
and can i use electronic shapes and these things there ?

ceremcem, 2021-01-16:
> and can i use electronic shapes and these things there ?

No, there is no "shape" support (yet). You'll need to define your mechanical layout with a DXF file,then snap your electronic components to appropriate positions: 

https://user-images.githubusercontent.com/6639874/104822036-05edd500-5851-11eb-8797-6852b9f36710.mp4

> I have to run it via command prompt (cmd) and then open Cygin and msys2 ?

There is no Cygwin involved in our instructions. You shouldn't need Cygwin. 

If you keep having troubles on Windows, I may try the instructions in a clean Windows installation.

ceremcem, 2021-03-17:
@nozge What is the latest status? Can we close the issue?

ceremcem, 2021-03-18:
Please feel free to open a new issue if the problem persists.

closed 2021-03-18 by ceremcem